vives-project-xp/UpgradeVerlichtingsysteem #14: Sprint review

RobbeYT, 2024-02-18:
Sprint review:
• Teams review the sprint
• Product owner is the only one who can approve stories as done
• Unaccepted or incomplete work is reviewed, prioritized and moved to another sprint
• Team agrees on what to demonstrate to stakeholders

RobbeYT, 2024-03-05:
Sprint was good veel progress gemaakt en al voorgesteld aan Arnold

Hebben vooral uitwerking gedaan en architectuur document gemaakt.

Ook powerpoint gemaakt en voorgesteld

als laatste ook een demo op verschillende gebouwen

PietKrobot, 2024-03-05:
research gedaan naar de verschillende mogelijke communicatie protocollen die we kunnen gebruiken. Welke apparatuur er vervolgens ook voor nodig zou zijn. Het maken van de README is ook compleet voor fase 1 en kan gemakkelijk verder aangevuld worden als dit nodig is in de toekomst.

RobbeYT, 2024-03-11:
[image]

goed gewerkt ;)